espressif/esp-insights #1: Insights SSID variable 1st letter uppercase

JohnMacrae, 2021-07-07:
Thanks for this functionality!

Using the switch example code and with variable and metrics enabled, I noticed that the SSID is reported with the 1st letter in uppercase.

in this case the ssid is
[image]
[image]

shubhamdp, 2021-07-07:
Hi @JohnMacrae this issue was fixed earlier today and you should be able to see the SSID as it is.

JohnMacrae, 2021-07-07:
thanks @shubhamdp - it may be that it has not been pushed since I installed the Insights only a few hours ago and there is no sign of anything new in the repo for the last few days.

closed 2021-07-07 by JohnMacrae