espressif/esp-insights #4: Idf patch not working

miroslavpetrov, 2021-11-05:
I am using esp idf 4.3.1 and trying to apply the patch for 4.3

Here is the output of the patch file:

```
    ~/esp/esp-idf  on  #v4.3.1 !8183  git apply -v /Users/miroslavpetrov/esp/esp-rainmaker/components/esp-insights/idf-patches/Diagnostics-support-in-esp-idf-tag-v4.3.patch
Checking patch components/espcoredump/CMakeLists.txt...
warning: components/espcoredump/CMakeLists.txt has type 100755, expected 100644
Checking patch components/espcoredump/Kconfig...
warning: components/espcoredump/Kconfig has type 100755, expected 100644
Hunk #1 succeeded at 90 (offset 8 lines).
Checking patch components/espcoredump/component.mk...
warning: components/espcoredump/component.mk has type 100755, expected 100644
Checking patch components/espcoredump/include/esp_core_dump.h...
warning: components/espcoredump/include/esp_core_dump.h has type 100755, expected 100644
error: while searching for:
 */
esp_err_t esp_core_dump_image_get(size_t* out_addr, size_t *out_size);

#ifdef __cplusplus
}
#endif

error: patch failed: components/espcoredump/include/esp_core_dump.h:89
error: components/espcoredump/include/esp_core_dump.h: patch does not apply
Checking patch components/espcoredump/include/port/riscv/esp_core_dump_summary_port.h...
Checking patch components/espcoredump/include/port/xtensa/esp_core_dump_summary_port.h...
Checking patch components/espcoredump/include_core_dump/esp_core_dump_port.h...
warning: components/espcoredump/include_core_dump/esp_core_dump_port.h has type 100755, expected 100644
Checking patch components/espcoredump/src/core_dump_elf.c...
warning: components/espcoredump/src/core_dump_elf.c has type 100755, expected 100644
Checking patch components/espcoredump/src/core_dump_flash.c...
warning: components/espcoredump/src/core_dump_flash.c has type 100755, expected 100644
error: while searching for:
#include "esp_flash_encrypt.h"
#include "esp_rom_crc.h"

const static DRAM_ATTR char TAG[] __attribute__((unused)) = "esp_core_dump_flash";

#if CONFIG_ESP_COREDUMP_ENABLE_TO_FLASH

error: patch failed: components/espcoredump/src/core_dump_flash.c:20
error: components/espcoredump/src/core_dump_flash.c: patch does not apply
Checking patch components/espcoredump/src/port/riscv/core_dump_port.c...
warning: components/espcoredump/src/port/riscv/core_dump_port.c has type 100755, expected 100644
Checking patch components/espcoredump/src/port/xtensa/core_dump_port.c...
warning: components/espcoredump/src/port/xtensa/core_dump_port.c has type 100755, expected 100644
    ~/esp/esp-idf  on  #v4.3.1 !8183 
```

shubhamdp, 2021-11-08:
@miroslavpetrov right now esp-insights [supports](https://github.com/espressif/esp-insights/tree/main/examples#set-up-esp-idf) release tags v4.3, v4.2.2, and v4.1.1. We'll add the support for v4.3.1 very soon.

miroslavpetrov, 2021-11-08:
Thanks for the info!

closed 2021-11-08 by miroslavpetrov

zavovi, 2022-06-03:
Fixed - missing spaces on some lines.