espressif/esp-insights #6: Unable to upload firmware package

tmedicci, 2022-02-13:
Sometimes it's difficult to upload a new firmware package. Most of the time it says "Something went wrong" without uploading the firmware properly. However, it seems that the firmware is uploaded besides the error message.

[image]

adwait-esp, 2022-02-14:
Is this issue intermittent? Able to upload the firmware package. Will keep a tab on this, we have raised an issue for the same.

tmedicci, 2022-02-14:
In fact, it's very frequent... I'd say that is more probably to show the error popup (even if the firmware is uploaded, by checking the uploaded firmware list) than showing a message that the firmware was uploaded successfully.

This screenshot, however, showed a moment that the firmware did not appear in the firmware list.

adwait-esp, 2022-02-17:
We have fixed the intermittent package upload failure issue and our deployed services are updated with the latest code changes. 

@tmedicci  can you please try dashboard.insights.espressif.com and check if you can upload your package now?

tmedicci, 2022-02-17:
Yes! I could upload the same package that failed before. Thanks!

closed 2022-02-17 by tmedicci