espressif/esp-insights #7: Unable to build

FanhuaCloud, 2022-05-18:
C:/Users/he/Desktop/esp-insights/components/esp_diagnostics/src/esp_diagnostics_utils.c: In function 'esp_diag_task_snapshot_get$part$0':
C:/Users/he/Desktop/esp-insights/components/esp_diagnostics/src/esp_diagnostics_utils.c:156:13: error: 'strncpy' output truncated before terminating nul copying as many bytes from a string as its length [-Werror=stringop-truncation]
             strncpy(tasks[i].name, name, strlen(name));
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

shubhamdp, 2022-05-19:
@FanhuaCloud thanks for putting up an issue, we'll release a fix sooner.

shubhamdp, 2022-05-27:
@FanhuaCloud [b53081cc0a362482f31584ed34fdde8db38220ef](https://github.com/espressif/esp-insights/commit/35457d230203ed21a37f29021dd542f1dc6ad0a9) will fix this issue. Can you please pull in the latest changes and try again.

FanhuaCloud, 2022-05-27:
> 

thanks, i will test it

FanhuaCloud, 2022-05-27:
> @FanhuaCloud [b53081cc0a362482f31584ed34fdde8db38220ef](https://github.com/espressif/esp-insights/commit/35457d230203ed21a37f29021dd542f1dc6ad0a9) will fix this issue. Can you please pull in the latest changes and try again.

thanks, i will test it

shubhamdp, 2022-06-06:
@FanhuaCloud If this is working for you can you please close this issue.

closed 2022-06-17 by FanhuaCloud