espressif/esp-insights #12: rtc_store info is used uninitialized in "rtc_store_read_complete"

brainstorm, 2022-10-12:
Hello,

I'm refreshing a ESP-IDF 4.4 project of mine towards ESP-IDF 5.x:

https://github.com/brainstorm/esp-modbus-power-meter

But I'm encountering some issues with `rtc_store` component in `esp-insights`, is this a known issue?:

```
(esp) rvalls@m1 esp-modbus-power-meter % idf.py build
Executing action: all (aliases: build)
Running ninja in directory /Users/rvalls/dev/personal/esp-modbus-power-meter/build
Executing "ninja all"...
[0/1] Re-running CMake...-- Component directory /Users/rvalls/esp/esp-idf/components/asio does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/rvalls/esp/esp-idf/components/cbor does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/rvalls/esp/esp-idf/components/coap does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/rvalls/esp/esp-idf/components/expat does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/rvalls/esp/esp-idf/components/libsodium does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/rvalls/esp/esp-idf/components/nghttp does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/rvalls/dev/personal/esp-modbus-power-meter/lib/esp-rainmaker/components/esp-insights does not contain a CMakeLists.txt file. No component will be added
-- Building ESP-IDF components for target esp32s2
Processing 1 dependencies:
[1/1] idf (5.1.0)-- Project sdkconfig file /Users/rvalls/dev/personal/esp-modbus-power-meter/sdkconfig
/Users/rvalls/dev/personal/esp-modbus-power-meter/sdkconfig:300 CONFIG_MCPWM_ISR_IN_IRAM was replaced with CONFIG_MCPWM_ISR_IRAM_SAFE
/Users/rvalls/dev/personal/esp-modbus-power-meter/sdkconfig:355 CONFIG_ESP32S2_DEFAULT_CPU_FREQ_80 was replaced with CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_80
/Users/rvalls/dev/personal/esp-modbus-power-meter/sdkconfig:356 CONFIG_ESP32S2_DEFAULT_CPU_FREQ_160 was replaced with CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_160
/Users/rvalls/dev/personal/esp-modbus-power-meter/sdkconfig:357 CONFIG_ESP32S2_DEFAULT_CPU_FREQ_240 was replaced with CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_240
/Users/rvalls/dev/personal/esp-modbus-power-meter/sdkconfig:358 CONFIG_ESP32S2_DEFAULT_CPU_FREQ_MHZ was replaced with CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ
/Users/rvalls/dev/personal/esp-modbus-power-meter/sdkconfig:376 CONFIG_ESP32S2_SPIRAM_SUPPORT was replaced with CONFIG_SPIRAM
/Users/rvalls/dev/personal/esp-modbus-power-meter/sdkconfig:379 CONFIG_ESP32S2_ULP_COPROC_ENABLED was replaced with CONFIG_ULP_COPROC_ENABLED
/Users/rvalls/dev/personal/esp-modbus-power-meter/sdkconfig:380 CONFIG_ESP32S2_ULP_COPROC_RESERVE_MEM was replaced with CONFIG_ULP_COPROC_RESERVE_MEM
/Users/rvalls/dev/personal/esp-modbus-power-meter/sdkconfig:381 CONFIG_ESP32S2_DEBUG_OCDAWARE was replaced with CONFIG_ESP_DEBUG_OCDAWARE
/Users/rvalls/dev/personal/esp-modbus-power-meter/sdkconfig:382 CONFIG_ESP32S2_BROWNOUT_DET was replaced with CONFIG_ESP_BROWNOUT_DET
/Users/rvalls/dev/personal/esp-modbus-power-meter/sdkconfig:383 CONFIG_ESP32S2_BROWNOUT_DET_LVL_SEL_7 was replaced with CONFIG_ESP_BROWNOUT_DET_LVL_SEL_7
/Users/rvalls/dev/personal/esp-modbus-power-meter/sdkconfig:384 CONFIG_ESP32S2_BROWNOUT_DET_LVL_SEL_6 was replaced with CONFIG_ESP_BROWNOUT_DET_LVL_SEL_6
/Users/rvalls/dev/personal/esp-modbus-power-meter/sdkconfig:385 CONFIG_ESP32S2_BROWNOUT_DET_LVL_SEL_5 was replaced with CONFIG_ESP_BROWNOUT_DET_LVL_SEL_5
/Users/rvalls/dev/personal/esp-modbus-power-meter/sdkconfig:386 CONFIG_ESP32S2_BROWNOUT_DET_LVL_SEL_4 was replaced with CONFIG_ESP_BROWNOUT_DET_LVL_SEL_4
/Users/rvalls/dev/personal/esp-modbus-power-meter/sdkconfig:387 CONFIG_ESP32S2_BROWNOUT_DET_LVL_SEL_3 was replaced with CONFIG_ESP_BROWNOUT_DET_LVL_SEL_3
/Users/rvalls/dev/personal/esp-modbus-power-meter/sdkconfig:388 CONFIG_ESP32S2_BROWNOUT_DET_LVL_SEL_2 was replaced with CONFIG_ESP_BROWNOUT_DET_LVL_SEL_2
/Users/rvalls/dev/personal/esp-modbus-power-meter/sdkconfig:389 CONFIG_ESP32S2_BROWNOUT_DET_LVL_SEL_1 was replaced with CONFIG_ESP_BROWNOUT_DET_LVL_SEL_1
/Users/rvalls/dev/personal/esp-modbus-power-meter/sdkconfig:390 CONFIG_ESP32S2_BROWNOUT_DET_LVL was replaced with CONFIG_ESP_BROWNOUT_DET_LVL
/Users/rvalls/dev/personal/esp-modbus-power-meter/sdkconfig:391 CONFIG_ESP32S2_TIME_SYSCALL_USE_RTC_FRC1 was replaced with CONFIG_NEWLIB_TIME_SYSCALL_USE_RTC_HRT
/Users/rvalls/dev/personal/esp-modbus-power-meter/sdkconfig:392 CONFIG_ESP32S2_TIME_SYSCALL_USE_RTC was replaced with CONFIG_NEWLIB_TIME_SYSCALL_USE_RTC
/Users/rvalls/dev/personal/esp-modbus-power-meter/sdkconfig:393 CONFIG_ESP32S2_TIME_SYSCALL_USE_FRC1 was replaced with CONFIG_NEWLIB_TIME_SYSCALL_USE_HRT
/Users/rvalls/dev/personal/esp-modbus-power-meter/sdkconfig:394 CONFIG_ESP32S2_TIME_SYSCALL_USE_NONE was replaced with CONFIG_NEWLIB_TIME_SYSCALL_USE_NONE
/Users/rvalls/dev/personal/esp-modbus-power-meter/sdkconfig:395 CONFIG_ESP32S2_RTC_CLK_SRC_INT_RC was replaced with CONFIG_RTC_CLK_SRC_INT_RC
/Users/rvalls/dev/personal/esp-modbus-power-meter/sdkconfig:396 CONFIG_ESP32S2_RTC_CLK_SRC_EXT_CRYS was replaced with CONFIG_RTC_CLK_SRC_EXT_CRYS
/Users/rvalls/dev/personal/esp-modbus-power-meter/sdkconfig:397 CONFIG_ESP32S2_RTC_CLK_SRC_EXT_OSC was replaced with CONFIG_RTC_CLK_SRC_EXT_OSC
/Users/rvalls/dev/personal/esp-modbus-power-meter/sdkconfig:398 CONFIG_ESP32S2_RTC_CLK_SRC_INT_8MD256 was replaced with CONFIG_RTC_CLK_SRC_INT_8MD256
/Users/rvalls/dev/personal/esp-modbus-power-meter/sdkconfig:399 CONFIG_ESP32S2_RTC_CLK_CAL_CYCLES was replaced with CONFIG_RTC_CLK_CAL_CYCLES
/Users/rvalls/dev/personal/esp-modbus-power-meter/sdkconfig:400 CONFIG_ESP32S2_NO_BLOBS was replaced with CONFIG_APP_NO_BLOBS
/Users/rvalls/dev/personal/esp-modbus-power-meter/sdkconfig:459 CONFIG_OTA_ALLOW_HTTP was replaced with CONFIG_ESP_HTTPS_OTA_ALLOW_HTTP
/Users/rvalls/dev/personal/esp-modbus-power-meter/sdkconfig:580 CONFIG_ESP_TASK_WDT was replaced with CONFIG_ESP_TASK_WDT_INIT
/Users/rvalls/dev/personal/esp-modbus-power-meter/sdkconfig:751 CONFIG_HAL_ASSERTION_SILIENT was replaced with CONFIG_HAL_ASSERTION_SILENT
-- App "esp-modbus-power-meter" version: portimao-27-g4491fac-dirty
-- Adding linker script /Users/rvalls/dev/personal/esp-modbus-power-meter/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /Users/rvalls/esp/esp-idf/components/esp_system/ld/esp32s2/sections.ld.in
-- Adding linker script /Users/rvalls/esp/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.ld
-- Adding linker script /Users/rvalls/esp/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.api.ld
-- Adding linker script /Users/rvalls/esp/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.libgcc.ld
-- Adding linker script /Users/rvalls/esp/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.newlib-funcs.ld
-- Adding linker script /Users/rvalls/esp/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.newlib-data.ld
-- Adding linker script /Users/rvalls/esp/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.spiflash.ld
-- Adding linker script /Users/rvalls/esp/esp-idf/components/soc/esp32s2/ld/esp32s2.peripherals.ld
ESP RainMaker Project commit: fee5190-dirty
ESP Insights Project commit: 
-- Components: app_insights app_reset app_trace app_update app_wifi bootloader bootloader_support bt cbor cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_common esp_diagnostics esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_insights esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_psram esp_rainmaker esp_ringbuf esp_rom esp_schedule esp_system esp_timer esp_wifi espcoredump esptool_py fatfs freertos gpio_button hal heap http_parser idf_test ieee802154 json json_generator json_parser log lwip mbedtls mdns mqtt newlib nvs_flash openthread partition_table perfmon protobuf-c protocomm pthread qrcode rmaker_common rtc_store sdmmc soc spi_flash spiffs tcp_transport tinyusb touch_element ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant ws2812_led xtensa
-- Component paths: /Users/rvalls/dev/personal/esp-modbus-power-meter/lib/esp-rainmaker/examples/common/app_insights /Users/rvalls/dev/personal/esp-modbus-power-meter/lib/esp-rainmaker/examples/common/app_reset /Users/rvalls/esp/esp-idf/components/app_trace /Users/rvalls/esp/esp-idf/components/app_update /Users/rvalls/dev/personal/esp-modbus-power-meter/lib/esp-rainmaker/examples/common/app_wifi /Users/rvall [... cut ...]
/Users/rvalls/dev/personal/esp-modbus-power-meter/lib/esp-rainmaker/components/esp-insights/components/rtc_store/src/rtc_store.c:120:14: note: 'info' declared here
  120 |     uint32_t info = rbuf_data->store->info;
      |              ^~~~
cc1: some warnings being treated as errors
[866/919] Building C object esp-idf/mdns/CMakeFiles/__idf_mdns.dir/mdns.c.objninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /Users/rvalls/dev/personal/esp-modbus-power-meter/build/log/idf_py_stderr_output_40444 and /Users/rvalls/dev/personal/esp-modbus-power-meter/build/log/idf_py_stdout_output_40444
```

brainstorm, 2022-10-12:
Some more esp-insights errors related to logging when the rtc_store_read_complete error above is circumvented:

```
% idf.py build
Executing action: all (aliases: build)
Running ninja in directory /Users/rvalls/dev/personal/esp-modbus-power-meter/build
Executing "ninja all"...
[1/1] cd /Users/rvalls/dev/personal/esp-modbus-power-meter/build/bootloader/esp-i.../Users/rvalls/dev/personal/esp-modbus-power-meter/build/bootloader/bootloader.binBootloader binary size 0x4f70 bytes. 0x2090 bytes (29%) free.
[13/21] Building C object esp-idf/esp_insights/CMakeFiles/__idf_esp_insights.dir/src/esp_insights.c.objFAILED: esp-idf/esp_insights/CMakeFiles/__idf_esp_insights.dir/src/esp_insights.c.obj 
/Users/rvalls/.espressif/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc -DESP_PLATFORM -DIDF_VER=\"v5.1-dev-1195-gfa5d0a3513\" -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/Users/rvalls/dev/personal/esp-modbus-power-meter/build/config -I/Users/rvalls/dev/personal/esp-modbus-power-meter/lib/esp-rainmaker/components/esp-insights/components/esp_insights/include -I/Users/rvalls/dev/personal/esp-modbus-power-meter/lib/esp-rainmaker/components/esp-insights/components/esp_insights/src -I/Users/rvalls/esp/esp-idf/components/newlib/platform_include -I/Users/rvalls/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/Users/rvalls/esp/esp-idf/components/freertos/esp_additions/include/freertos -I/Users/rvalls/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/Users/rvalls/esp/esp-idf/components/freertos/esp_additions/include -I/Users/rvalls/esp/esp-idf/components/esp_hw_support/include -I/Users/rvalls/esp/esp-idf/components/esp_hw_support/include/soc -I/Users/rvalls/esp/esp-idf/components/esp_hw_support/include/soc/esp32s2 -I/Users/rvalls/esp/esp-idf/components/esp_hw_support/port/esp32s2/. -I/Users/rvalls/esp/esp-idf/components/esp_hw_support/port/esp32s2/private_include -I/Users/rvalls/esp/esp-idf/components/heap/include -I/Users/rvalls/esp/esp-idf/components/log/include -I/Users/rvalls/esp/esp-idf/components/soc/include -I/Users/rvalls/esp/esp-idf/components/soc/esp32s2/. -I/Users/rvalls/esp/esp-idf/components/soc/esp32s2/include -I/Users/rvalls/esp/esp-idf/components/hal/esp32s2/include -I/Users/rvalls/esp/esp-idf/components/hal/include -I/Users/rvalls/esp/esp-idf/components/hal/platform_port/include -I/Users/rvalls/esp/esp-idf/components/esp_rom/include -I/Users/rvalls/esp/esp-idf/components/esp_rom/include/esp32s2 -I/Users/rvalls/esp/esp-idf/components/esp_rom/esp32s2 -I/Users/rvalls/esp/esp-idf/components/esp_common/include -I/Users/rvalls/esp/esp-idf/components/esp_system/include -I/Users/rvalls/esp/esp-idf/components/esp_system/port/soc -I/Users/rvalls/esp/esp-idf/components/esp_system/port/include/private -I/Users/rvalls/esp/esp-idf/components/xtensa/include -I/Users/rvalls/esp/esp-idf/components/xtensa/esp32s2/include -I/Users/rvalls/esp/esp-idf/components/lwip/include -I/Users/rvalls/esp/esp-idf/components/lwip/include/apps -I/Users/rvalls/esp/esp-idf/components/lwip/include/apps/sntp -I/Users/rvalls/esp/esp-idf/components/lwip/lwip/src/include -I/Users/rvalls/esp/esp-idf/components/lwip/port/esp32/include -I/Users/rvalls/esp/esp-idf/components/lwip/port/esp32/include/arch -I/Users/rvalls/dev/personal/esp-modbus-power-meter/lib/esp-rainmaker/components/esp-insights/components/esp_diagnostics/include -I/Users/rvalls/dev/personal/esp-modbus-power-meter/lib/esp-rainmaker/components/esp-insights/components/cbor/port/include -I/Users/rvalls/dev/personal/esp-modbus-power-meter/lib/esp-rainmaker/components/esp-insights/components/rmaker_common/include -I/Users/rvalls/esp/esp-idf/components/esp_event/include -I/Users/rvalls/esp/esp-idf/components/espcoredump/include -I/Users/rvalls/esp/esp-idf/components/espcoredump/include/port/xtensa -I/Users/rvalls/dev/personal/esp-modbus-power-meter/lib/esp-rainmaker/components/esp-insights/components/rtc_store/include -I/Users/rvalls/esp/esp-idf/components/nvs_flash/include -I/Users/rvalls/esp/esp-idf/components/spi_flash/include -I/Users/rvalls/esp/esp-idf/components/esp_timer/include -I/Users/rvalls/esp/esp-idf/components/esp_wifi/include -I/Users/rvalls/esp/esp-idf/components/esp_phy/include -I/Users/rvalls/esp/esp-idf/components/esp_phy/esp32s2/include -I/Users/rvalls/esp/esp-idf/components/esp_netif/include -mlongcalls  -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -Os -freorder-blocks -fmacro-prefix-map=/Users/rvalls/dev/personal/esp-modbus-power-meter=. -fmacro-prefix-map=/Users/rvalls/esp/esp-idf=/IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1 -std=gnu17 -Wno-old-style-declaration -MD -MT esp-idf/esp_insights/CMakeFiles/__idf_esp_insights.dir/src/esp_insights.c.obj -MF esp-idf/esp_insights/CMakeFiles/__idf_esp_insights.dir/src/esp_insights.c.obj.d -o esp-idf/esp_insights/CMakeFiles/__idf_esp_insights.dir/src/esp_insights.c.obj -c /Users/rvalls/dev/personal/esp-modbus-power-meter/lib/esp-rainmaker/components/esp-insights/components/esp_insights/src/esp_insights.c
In file included from /Users/rvalls/dev/personal/esp-modbus-power-meter/lib/esp-rainmaker/components/esp-insights/components/esp_insights/src/esp_insights.c:15:
[... cut ...]
ninja: build stopped: subcommand failed.
HINT: The issue is better to resolve by replacing format specifiers to 'PRI'-family macros (include <inttypes.h> header file).
ninja failed with exit code 1, output of the command is in the /Users/rvalls/dev/personal/esp-modbus-power-meter/build/log/idf_py_stderr_output_45669 and /Users/rvalls/dev/personal/esp-modbus-power-meter/build/log/idf_py_stdout_output_45669
```

After `#if 0`-ing that too, seems like I'm out of the weeds, but it'd be nice to have those two fixed upstream ;)

brainstorm, 2022-10-12:
Thanks @vikramdattu 

> For error from `rtc_store`, unfortunately I am not able to re-produce this. Can you please check if you have updated your IDF `release/v5.0` branch?
> Can you please setup IDF5.0 tools in new terminal instead of in the one where release/v4.4 tools setup?

I'm actually using `master`, with commit `fa5d0a351343b45ee06d221b7c29028672a4c3c2`, can you test it against HEAD?

vikramdattu, 2022-10-12:
@brainstorm I do not get error even with master! :(

Can you please help me with below patch? This should fix the issue for good. 
[rtc_store_info.txt](https://github.com/espressif/esp-insights/files/9765429/rtc_store_info.txt)

brainstorm, 2022-10-12:
Weird, I switched back to master and now that error does not appear :-S

Anyway, I've applied your patch and indeed solves the problem, thanks!

```
(base) rvalls@m1 esp-insights % patch -p1 < rtc_store_info.txt
patching file components/rtc_store/src/rtc_store.c
Hunk #3 succeeded at 417 (offset 4 lines).
```

Now I have to figure out a way to have platformio and esp-idf to co-exist: https://github.com/espressif/esp-idf/issues/9760#issuecomment-1276018599... closing this one, push that patch of yours on main on this repo, please!

brainstorm, 2022-10-13:
So... @vikramdattu, are you pushing that patch to `main`? :)

vikramdattu, 2022-10-14:
It's already done internally. It shall make it's way to github today by EOD. Apologies for the delay.

brainstorm, 2022-10-24:
@vikramdattu I believe there's still something off with those RTC routines from ESP Insights. I just got this compiling my project with your latest changes:

```
% md5sum lib/esp-rainmaker/components/esp-insights/components/rtc_store/src/rtc_store.c
9a31af0588331f90b1fe3600b3fb0395  lib/esp-rainmaker/components/esp-insights/components/rtc_store/src/rtc_store.c
```

and PlatformIO fails at compile time:

```
(...)
Compiling .pio/build/esp32-s2-saola-1/sdmmc/sdmmc_common.o
Archiving .pio/build/esp32-s2-saola-1/esp-idf/qrcode/libqrcode.a
Compiling .pio/build/esp32-s2-saola-1/sdmmc/sdmmc_init.o
Indexing .pio/build/esp32-s2-saola-1/esp-idf/qrcode/libqrcode.a
lib/esp-rainmaker/components/esp-insights/components/rtc_store/src/rtc_store.c: In function 'rtc_store_non_critical_data_write':
lib/esp-rainmaker/components/esp-insights/components/rtc_store/src/rtc_store.c:251:44: error: incompatible type for argument 1 of 'rtc_store_read_complete'
         rtc_store_read_complete(s_priv_data.non_critical, sizeof(header) + header.len);
                                 ~~~~~~~~~~~^~~~~~~~~~~~~
lib/esp-rainmaker/components/esp-insights/components/rtc_store/src/rtc_store.c:121:50: note: expected 'rbuf_data_t *' {aka 'struct <anonymous> *'} but argument is of type 'rbuf_data_t' {aka 'struct <anonymous>'}
 static void rtc_store_read_complete(rbuf_data_t *rbuf_data, size_t len)
                                     ~~~~~~~~~~~~~^~~~~~~~~
Compiling .pio/build/esp32-s2-saola-1/sdmmc/sdmmc_io.o
*** [.pio/build/esp32-s2-saola-1/lib/esp-rainmaker/components/esp-insights/components/rtc_store/src/rtc_store.o] Error 1
============================================================= [FAILED] Took 40.79 seconds =============================================================
```

With ESP-IDF 5.x it compiles fine, but there are quite a few runtime issues with it (see **rtc_store_data_critical_data_write** errors in white text):

[image]

The Resource allocation failure errors in red are unrelated and already reported in https://github.com/espressif/esp-modbus/issues/10.

brainstorm, 2022-10-27:
Commenting the offending code for now locally, just in case somebody else is facing the same issues:

```diff
diff --git a/components/rtc_store/src/rtc_store.c b/components/rtc_store/src/rtc_store.c
index a744fae..75625ef 100644
--- a/components/rtc_store/src/rtc_store.c
+++ b/components/rtc_store/src/rtc_store.c
@@ -242,22 +242,22 @@ esp_err_t rtc_store_non_critical_data_write(const char *dg, void *data, size_t l
         return ESP_FAIL;
     }
 
-#if CONFIG_RTC_STORE_OVERWRITE_NON_CRITICAL_DATA
-    data_store_info_t *info = (data_store_info_t *) &s_priv_data.non_critical.store->info;
-    /* Make enough room for the item */
-    while (data_store_get_free(s_priv_data.non_critical.store) < req_free) {
-        uint8_t *read_ptr = s_priv_data.non_critical.store->buf + info->read_offset;
-        memcpy(&header, read_ptr, sizeof(header));
-        rtc_store_read_complete(s_priv_data.non_critical, sizeof(header) + header.len);
-    }
-#else // just check if we have enough space to write the item
-    curr_free = data_store_get_free(s_priv_data.non_critical.store);
-    if (curr_free < req_free) {
-        xSemaphoreGive(s_priv_data.non_critical.lock);
-        esp_event_post(RTC_STORE_EVENT, RTC_STORE_EVENT_NON_CRITICAL_DATA_LOW_MEM, NULL, 0, 0);
-        return ESP_ERR_NO_MEM;
-    }
-#endif
+// #if CONFIG_RTC_STORE_OVERWRITE_NON_CRITICAL_DATA
+//     data_store_info_t *info = (data_store_info_t *) &s_priv_data.non_critical.store->info;
+//     /* Make enough room for the item */
+//     while (data_store_get_free(s_priv_data.non_critical.store) < req_free) {
+//         uint8_t *read_ptr = s_priv_data.non_critical.store->buf + info->read_offset;
+//         memcpy(&header, read_ptr, sizeof(header));
+//         rtc_store_read_complete(s_priv_data.non_critical, sizeof(header) + header.len);
+//     }
+// #else // just check if we have enough space to write the item
+//     curr_free = data_store_get_free(s_priv_data.non_critical.store);
+//     if (curr_free < req_free) {
+//         xSemaphoreGive(s_priv_data.non_critical.lock);
+//         esp_event_post(RTC_STORE_EVENT, RTC_STORE_EVENT_NON_CRITICAL_DATA_LOW_MEM, NULL, 0, 0);
+//         return ESP_ERR_NO_MEM;
+//     }
+// #endif
```

Unsure about side effects this might have, but I need to keep moving for now ;)

vikramdattu, 2022-10-27:
Hi @brainstorm this too has been addressed and will be available.

The change below should fix it:
-rtc_store_read_complete(s_priv_data.non_critical, sizeof(header) + header.len);
+rtc_store_read_complete(&s_priv_data.non_critical, sizeof(header) + header.len);

brainstorm, 2022-10-27:
> Hi @brainstorm this too has been addressed and will be available.
> 
> The change below should fix it: -rtc_store_read_complete(s_priv_data.non_critical, sizeof(header) + header.len); +rtc_store_read_complete(&s_priv_data.non_critical, sizeof(header) + header.len);

Thanks @vikramdattu! Let me know when it's upstreamed and I'll pull the changes in & close this issue for good ;)

vikramdattu, 2023-04-04:
Hi @brainstorm missed to update on this. The fix is now a part of main branch. You you haven't yet, you may go ahead and pull the main branch.

closed 2023-04-04 by shahpiyushv