espressif/esp-insights #15: Not able to install the zip library in new Arduino 2.0

rajsimiot, 2022-11-07:
I have downloaded the zip library but not able to install it in the Arduino 2.0. Getting error message library not valid.

sanketwadekar, 2022-11-08:
Hi @rajsimiot,
Follow the steps mentioned in the [documentation website](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#installing) to install ESP32 Arduino.

shahpiyushv, 2022-11-08:
@rajsimiot , just curious to know which zip file you mean here and if you are trying to do something with Insights on Arduino? For Arduino specific issues not related to ESP Insights, the appropriate project is [arduino-esp32](https://github.com/espressif/arduino-esp32/issues)

sanketwadekar, 2023-01-02:
Hi @rajsimiot
ESP-Insights library has been added to Arduino-ESP32 in version 2.0.6. Please download the latest release from the Arduino IDE.

closed 2023-04-04 by shahpiyushv