espressif/esp-insights #16: Failed to resolve component 'rmaker_common'.

Kampi, 2022-11-17:
I try to add ESP Insights to an ESP32-C3 project. I have added `esp_diagnostics`, `esp_insights` and `esp-rainmaker` (git clone --recursive https://github.com/espressif/esp-rainmaker.git) to the `components` directory of my project. During the build process the following error is thrown:

```
PS C:\Projects\wifi-prov> pio run -t menuconfig
Processing debug (board: esp32-c3-devkitm-1; platform: https://git.traplinked.com/traplinked/hardware/platformio/platforms/platform-espidf.git#idf_4.4.2; framework: espidf)
------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c3-devkitm-1.html
PLATFORM: Espressif 32 (4.4.2+sha.ce6cdd0) > Espressif ESP32-C3-DevKitM-1
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf @ 3.40402.0+sha.67bfcf25
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 1.22851.191205 (2.28.51)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch3
Reading CMake configuration...
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.38.1.windows.1")
-- Component directory C:/Projects/wifi-prov/components/esp-rainmaker does not contain a CMakeLists.txt file. No component will be added
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/Daniel/.platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc.exe
-- Check for working C compiler: C:/Users/Daniel/.platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc.exe
-- Check for working C compiler: C:/Users/Daniel/.platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Users/Daniel/.platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-g++.exe
-- Check for working CXX compiler: C:/Users/Daniel/.platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32c3
-- Configuring incomplete, errors occurred!
See also "C:/Projects/wifi-prov/.pio/build/debug/CMakeFiles/CMakeOutput.log".

CMake Error at C:/Users/Daniel/.platformio/packages/framework-espidf@src-7f04cf235f256dbc46d61361b84e2ab6/tools/cmake/build.cmake:201 (message):
  Failed to resolve component 'rmaker_common'.
Call Stack (most recent call first):
  C:/Users/Daniel/.platformio/packages/framework-espidf@src-7f04cf235f256dbc46d61361b84e2ab6/tools/cmake/build.cmake:232 (__build_resolve_and_add_req)
  C:/Users/Daniel/.platformio/packages/framework-espidf@src-7f04cf235f256dbc46d61361b84e2ab6/tools/cmake/build.cmake:509 (__build_expand_requirements)
  C:/Users/Daniel/.platformio/packages/framework-espidf@src-7f04cf235f256dbc46d61361b84e2ab6/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:4 (project)
  ```

How can I fix this?

vikramdattu, 2023-01-02:
Hello @Kampi sorry for the late reply. Did you get resolution for the issue?

ESP-Insights needs to clone `rmaker_common` on it's own. Simply run `git submodule update --init --recursive` inside `esp-insights` clone and the error should go away.

Kampi, 2023-01-03:
No it doesn´t work :(

vikramdattu, 2023-01-04:
@Kampi

> I try to add ESP Insights to an ESP32-C3 project. I have added esp_diagnostics, esp_insights and esp-rainmaker (git clone --recursive https://github.com/espressif/esp-rainmaker.git) to the components directory of my project.

I think there is an issue with how you are trying to include the project.
First of all, you do not need to do explicit clone/copy of `esp_insights` and `esp_diagnostics` components to your project directory, as `esp-insights` project is already is a submodule to `esp-rainmaker` and doing recursive clone (or `git submodule --init --recursive` later if you forgot the recursive clone) would just do.

Please refer this example for how to include the esp-insights with rainmaker: https://github.com/espressif/esp-rainmaker/blob/fe94cc6994c4d012bad3ce04d6ae07b616894714/examples/led_light/CMakeLists.txt#L12

Kampi, 2023-01-04:
Sorry, but I don´t get it...
I execute the following steps:
- Create a new project in VS Code by using PlatformIO (using `esp-idf`)
- Create `components` directory
- `cd components`
- `git clone https://github.com/espressif/esp-insights.git`
- `cd esp-insights`
- `git submodule update --init --recursive`

[image]

I copy the CMake directives from the example to `src`CMakeLists.txt`:
```
set(INSIGHTS_PATH ${CMAKE_CURRENT_LIST_DIR})
set(EXTRA_COMPONENT_DIRS ${INSIGHTS_PATH}/components)
```

And I get the error `Error: Couldn't find the main target of the project!`.

closed 2024-02-05 by Kampi