espressif/esp-insights #17: "Please provide a valid package name"

ivancmz, 2022-11-29:
When trying to add the Firmware Package on the dashboard I get the "Please provide a valid package name" error.

[image]

The Project name is: R84.24 
The version is:  2_0_0_221118
The zip file generated is: R84.24-v2_0_0_221118.zip
The package name I'm using is: R84.24-v2_0_0_221118, also tried R84.24-2_0_0_221118 and other combinations.

[image]

ivancmz, 2022-11-30:
Changed version to a more "traditional" scheme (i.e., R84.24-2.0), same issue.

Changed project name to R84_24 to avoid dots (i.e., R84_24-2.0), same issue (R84_24-2.0).

Changed project name to Light to match exactly the dialogs example values (Light-2.0) and it still fails.

[image]

ivancmz, 2022-11-30:
It finally worked after removing the version part, that is package name "Light" without the "-2.0"

shahpiyushv, 2022-11-30:
@ivancmz , good to know you could get it working but we do understand that this is very inconvenient. The package name can actually be any friendly name that the user wants, independent of actual project name and version. We will fix this bug asap.

ivancmz, 2022-11-30:
> @ivancmz , good to know you could get it working but we do understand that this is very inconvenient. The package name can actually be any friendly name that the user wants, independent of actual project name and version. We will fix this bug asap.

Thanks @shahpiyushv. What I've seen is that it only allows letters in the name, no numbers or special characters like underscore. I see now that I can use any name, but since the placeholder example was "Light-2.0" I assumed it had to match the project name and version.

adwait-esp, 2022-12-13:
This issue was caused by some regression, it has been fixed and pushed to production. @ivancmz can you please check and close if the issue is fixed?

closed 2023-01-24 by ivancmz