espressif/esp-insights #19: can link

haoshijing, 2022-12-02:
undefined reference to `esp_insights_init'

haoshijing, 2022-12-02:
Linking CXX executable xinfengji.elf
/Users/mac/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/11.2.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/main/libmain.a(main.c.obj):(.literal.app_main+0x40): undefined reference to `esp_insights_init'
/Users/mac/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/11.2.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/main/libmain.a(main.c.obj): in function `app_main':
/Users/mac/Downloads/baozhi/main/main.c:2730: undefined reference to `esp_insights_init'
collect2: error: ld returned 1 exit status
make[2]: *** [xinfengji.elf] Error 1
make[1]: *** [CMakeFiles/xinfengji.elf.dir/all] Error 2
make: *** [all] Error 2
make failed with exit code 2, output of the command is in the /Users/mac/Downloads/baozhi/build/log/idf_py_stderr_output_51754 and /Users/mac/Downloads/baozhi/build/log/idf_py_stdout_output_51754

vikramdattu, 2022-12-02:
Hi @haoshijing can you please enable ESP-Insights via menuconfig? You can do this as follows:
`idf.py menuconfig` > `component config` > `ESP Insights`

closed 2022-12-03 by haoshijing