espressif/esp-insights #20: can send data to cloud error will case device restart

haoshijing, 2022-12-05:
05/12/22 21:01:46
Software reset due to exception/panic (ESP_RST_PANIC)

ERROR
05/12/22 21:02:39
E (7905) esp-tls: couldn't get hostname for :client.insights.espressif.com: getaddrinfo() returns 202, addrinfo=0x0

ERROR
05/12/22 21:02:39
E (7925) esp-tls: Failed to open new connection

ERROR
05/12/22 21:02:39
E (7925) transport_base: Failed to open a new connection

ERROR
05/12/22 21:02:39
E (7935) HTTP_CLIENT: Connection failed, sock < 0

ERROR
05/12/22 21:02:39
E (7935) tport_https: esp_http_client_perform failed err:0x7002

ERROR
05/12/22 21:02:39
E (7945) esp-tls: couldn't get hostname for :client.insights.espressif.com: getaddrinfo() returns 202, addrinfo=0x0

ERROR
05/12/22 21:02:39
E (7955) esp-tls: Failed to open new connection

ERROR
05/12/22 21:02:39
E (7965) transport_base: Failed to open a new connection

ERROR
05/12/22 21:02:39
E (7965) HTTP_CLIENT: Connection failed, sock < 0

ERROR
05/12/22 21:02:39
E (7975) tport_https: esp_http_client_perform failed err:0x700

vikramdattu, 2022-12-07:
Hi @haoshijing can you please provide a small example to reproduce the issue?

haoshijing, 2022-12-09:
can you give me a email address , i can give my total project code to you

haoshijing, 2022-12-15:
addr2line -e /opt/esp32newwind/build/xinfengji.elf  -a -f 0x4037a422:0x3fcb8350 0x40383c2d:0x3fcb8370 0x4038766a:0x3fcb8390 0x40385c7d:0x3fcb8400 0x40383ce4:0x3fcb8420 0x40383cda:0x00000000
0x4037a422
panic_abort
/root/esp/esp-idf/components/esp_system/panic.c:412
0x40383c2d
esp_system_abort
/root/esp/esp-idf/components/esp_system/esp_system.c:135
0x4038766a
vApplicationStackOverflowHook
/root/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:388
0x40385c7d
vTaskSwitchContext
/root/esp/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:3647
0x40383ce4
_frxt_dispatch
/root/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:418
0x40383cda
_frxt_int_exit
/root/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:213

vikramdattu, 2023-01-02:
@haoshijing the fix for the issue is now public.  Can you please pull the latest ESP-insights and check?

closed 2023-04-04 by shahpiyushv