espressif/esp-insights #24: ESP Insights examples (https case)- No logs in Dashboard

Sky-Soo-Ap, 2023-05-18:
I tried these 2 examples “minimal_diagnostics” and “diagnostics_smoke_test” with https case, but no logs in Dashboard web-site.
esp-rainmaker version is “cc861bf” and ESP-IDF version is v5.0.1.

vikramdattu, 2023-05-18:
Hello @Sky-Soo-Ap can you please check the time range selected on the dashboard?

Also, please do share the screenshot of the same. You may enable Insights debug prints (Option is present under `idf.py menuconfig > components config > ESP Insights`) and attach the console logs here.

Sky-Soo-Ap, 2023-05-18:
Hi vikramdattu,
I enabled Insights debug prints still no logs in Dashboard.
I’m attaching the log file, sdkconfig file and Dashboard screenshots.
[ESP_Insights_Example_No_Logs_On_Dashboard.log](https://github.com/espressif/esp-insights/files/11504901/ESP_Insights_Example_No_Logs_On_Dashboard.log)
[sdkconfig.log](https://github.com/espressif/esp-insights/files/11504918/sdkconfig.log)
[image]
[image]

vikramdattu, 2023-05-18:
Hi @Sky-Soo-Ap from top right corner of the dashboard, please select a wider time range for data to be displayed. Hope this helps.

Sky-Soo-Ap, 2023-05-18:
Hi vikramdattu,
Oh, I see those logs, thanks and I closed it.
BTW, could you set the default Dashboard for showing today logs?

closed 2023-05-18 by Sky-Soo-Ap

vikramdattu, 2023-05-18:
@Sky-Soo-Ap this does actually happen for a month time period and the range keep updating every 5 minutes and needs a refresh of dashboard.
We will consider your feedback and explore ways to improve the experience.

Sky-Soo-Ap, 2023-05-18:
Hi vikramdattu,
A month period is okay;
But, the End time "10:10:12" seems no change, logout then login again, still is "10:10:12";
May be can set it to End of today "23:59:59", then user login will see the logs.