espressif/esp-insights #25: ESP Insights - MQTT case - No Error, Warning and Event logs in Dashboard.

Sky-Soo-Ap, 2023-05-19:
Hi Dear,
I tried temperature_sensor example and added a smoke test from diagnostics_smoke_test example.
Following is the attached log file, sdkconfig file and Dashboard screenshot.
[ESP_Insights_MQTT_case_No_Error_Warning_Event_logs.log](https://github.com/espressif/esp-insights/files/11513918/ESP_Insights_MQTT_case_No_Error_Warning_Event_logs.log)
[sdkconfig.log](https://github.com/espressif/esp-insights/files/11513920/sdkconfig.log)
[image]

vikramdattu, 2023-05-19:
@Sky-Soo-Ap the  smoke_test example by design crashes randomly for 5 times. Hence the crashes.
If you reboot manually, again the device will go into 5 times crash mode.

The line introducing the crash: https://github.com/espressif/esp-insights/blob/d8b2dfcff850a0b6ae44c406d5f40a3c9364dc8c/examples/diagnostics_smoke_test/main/app_main.c#L73

vikramdattu, 2023-05-19:
@Sky-Soo-Ap can you share the logs with Insights debug enabled, like you did in another issue?

Sky-Soo-Ap, 2023-05-22:
@[vikramdattu](https://github.com/vikramdattu)
Following logs with Insights debug enabled.
[ESP_Insights_MQTT_case_No_Error_Warning_Event_logs_1.log](https://github.com/espressif/esp-insights/files/11526095/ESP_Insights_MQTT_case_No_Error_Warning_Event_logs_1.log)
>> did you make any changes to config to the Insight enable API call?
No, following is the source code (app_main.txt) and config (config.log).
[sdkconfig.log](https://github.com/espressif/esp-insights/files/11526097/sdkconfig.log)
[app_main.txt](https://github.com/espressif/esp-insights/files/11526102/app_main.txt)
Did you check if you are getting metrics/variables data on the dashboard?
No, following is the screenshot on Dashboard.
[image]

vikramdattu, 2023-05-23:
Hello @Sky-Soo-Ap,

There are some change in the log format, which is causing this. 

e.g. in your case
```
"\u001b[0;33mW (36229) app_main: [count][3]\u001b[0m\n"
```
this entire string is getting treated as an argument to printf function for some reason!
Do you by any chance have changes in IDF, especially about log output which may cause this? Do check for the same.

As far as, metrics and variables are concerned, those should still be visible on he dashboard. Can you please click on Metrics/variables tabs and check if those are showing?

Sky-Soo-Ap, 2023-05-23:
Hi @[vikramdattu](https://github.com/vikramdattu),
Metrics tab has just a little values and, Variables tab is empty on Dashboard.
[image]
[image]