espressif/esp-insights #27: How to change node id

antoniuschan99, 2023-07-26:
Hello, how do I change the node id that is being sent to insights? If not possible, is it using the mac address as the node id? Also, I see a node id in the rainmaker component. Thanks!

shahpiyushv, 2023-07-26:
@antoniuschan99 , while using Insights over HTTP, the mac address gets used as the node id by default. You can override it by setting a custom node id in [esp_insights_config_t](https://github.com/espressif/esp-insights/blob/56eec5c29a855bb019d49eb1c46ba5b3d389d561/components/esp_insights/include/esp_insights.h#L42C3-L42C24) which is passed to [esp_insights_init()](https://github.com/espressif/esp-insights/blob/56eec5c29a855bb019d49eb1c46ba5b3d389d561/components/esp_insights/include/esp_insights.h#L141).

The node id in RainMaker, however, cannot be customised. It is set to mac addeess for self claiming and a randomly assigned uuid for assisted/host claiming. The same gets used for Insights over MQTT.

antoniuschan99, 2023-07-26:
ok great thanks!

closed 2023-07-26 by antoniuschan99