espressif/esp-insights #28: Reset Reason

antoniuschan99, 2023-07-27:
In the esp insights marketing, it says you have esp insights upload the core dump logs to the cloud if there was a crash. Is there an esp insights implementation or should we use esp_reset_reason()

`#include "esp_system.h"

void app_main() {
    // Check the reason for the last reset
    esp_reset_reason_t reset_reason = esp_reset_reason();

    // Determine the reset reason and take appropriate action
    switch(reset_reason) {
        case ESP_RST_UNKNOWN:
            // This is the power-on reset, and the reason is unknown.
            // It could be a fresh boot or due to power-up.
            break;

        case ESP_RST_POWERON:
            // The system was reset due to a power-on event.
            break;

        case ESP_RST_EXT:
            // The system was reset by an external reset signal.
            break;

        case ESP_RST_SW:
            // The system was reset by software using the RESET button.
            break;

        case ESP_RST_PANIC:
            // The system was reset due to a panic (exception/crash).
            // You can take necessary recovery or logging actions here.
            break;

        case ESP_RST_INT_WDT:
            // The system was reset because the internal watchdog timer timed out.
            break;

        case ESP_RST_TASK_WDT:
            // The system was reset because a task watchdog timeout occurred.
            break;

        case ESP_RST_WDT:
            // The system was reset by the hardware watchdog timer.
            break;

        case ESP_RST_DEEPSLEEP:
            // The system was reset from deep sleep mode.
            break;

        default:
            // Other reset reasons, if any, can be handled here.
            break;
    }

    // Your application code goes here...
}
`

vikramdattu, 2023-07-28:
@antoniuschan99 
ESP-Insights already sends the reboot reason. 
It also optionally sends the `coredump` on crash. You need to enable IDF's `coredump` feature to be able to do the same. Please follow this and set core dump to flash if you want to get crash info: https://docs.espressif.com/projects/esp-idf/en/release-v5.1/esp32/api-guides/core_dump.html

antoniuschan99, 2023-07-28:
Thanks for your reply. I mean upon reboot, is there an implementation/function within esp insights api to know if the reboot was caused by a crash? Yes, I am currently able to get it to send coredump info to esp insights cloud.

vikramdattu, 2023-07-28:
@antoniuschan99 as you've suggested in main issue description, `esp_reset_reason` is the correct way to go. The same API is also used by ESP-Insights internally.

antoniuschan99, 2023-07-28:
ok great thanks!

closed 2023-07-28 by antoniuschan99