espressif/esp-insights #29: Custom metrics demo

ftc300, 2023-07-28:
I would like to inquire about monitoring the Bluetooth connection status and how to create a custom Metric for this purpose. Could you please provide me with a demo or example on how to achieve this?

closed 2023-08-01 by ftc300

vikramdattu, 2023-08-01:
@ftc300 I hope you figured this out.
We will add an example that will showcase a use case with the usage of custom metrics real soon.

dolphin22, 2023-08-10:
@vikramdattu please kindly spend time showing us how to add custom metrics and variables as well. Thank you in advanced!

vikramdattu, 2023-08-10:
Hi @dolphin22 adding a custom metric/veriable has two parts:
1. Register a metrics/veriable // one time at the start after initialising insights
2. Send the data // this can be multiple times as per requirement

```
#include <esp_diagnostics_metric.h>`
#define TAG "heap"
#define KEY "free"
#define LABEL "amount of free heap"
#define PATH "Heap.Free"
```
Register a metric: 
`esp_diag_metrics_register(TAG, KEY, LABEL, PATH ESP_DIAG_DATA_TYPE_UINT);` // [API ref](https://github.com/espressif/esp-insights/blob/d082711e0114e4750fa670d6b755f44611c8ab33/components/esp_diagnostics/include/esp_diagnostics_metrics.h#L82)

Send the data: (this can be done multiple times)

```
uint32_t free_heap = <code to get free heap>
esp_diag_metrics_add_uint(KEY,  free heap); 
``` 

[API ref](https://github.com/espressif/esp-insights/blob/d082711e0114e4750fa670d6b755f44611c8ab33/components/esp_diagnostics/include/esp_diagnostics_metrics.h#L163)

Hope this helps!

dolphin22, 2023-08-10:
> Hi @dolphin22 adding a custom metric/veriable has two parts:
> 
> 1. Register a metrics/veriable // one time at the start after initialising insights
> 2. Send the data // this can be multiple times as per requirement
> 
> ```
> #include <esp_diagnostics_metric.h>`
> #define TAG "heap"
> #define KEY "free"
> #define LABEL "amount of free heap"
> #define PATH "Heap.Free"
> ```
> 
> Register a metric: `esp_diag_metrics_register(TAG, KEY, LABEL, PATH ESP_DIAG_DATA_TYPE_UINT);` // [API ref](https://github.com/espressif/esp-insights/blob/d082711e0114e4750fa670d6b755f44611c8ab33/components/esp_diagnostics/include/esp_diagnostics_metrics.h#L82)
> 
> Send the data: (this can be done multiple times)
> 
> ```
> uint32_t free_heap = <code to get free heap>
> esp_diag_metrics_add_uint(KEY,  free heap); 
> ```
> 
> [API ref](https://github.com/espressif/esp-insights/blob/d082711e0114e4750fa670d6b755f44611c8ab33/components/esp_diagnostics/include/esp_diagnostics_metrics.h#L163)
> 
> I have shown the example to add metrics, you can extend this similarly for variables. Hope this helps!

It works, thanks!