espressif/esp-insights #30: How to track an ESP Insights Event

antoniuschan99, 2023-08-03:
I am following minimal diagnostics example and in my code after sending the esp insights data, it will go to deep sleep. I noticed I have to do something like this in order to get the data on Esp Insights Cloud. Thank you!

`
  esp_diag_heap_metrics_dump();
  vTaskDelay(5000 / portTICK_PERIOD_MS);
  //deep_sleep code
`

Looking at esp_insights.h on line 52 I see this
```
/** Asynchronous data send succeded. Event data contains the msg_id of the data. */
INSIGHTS_EVENT_TRANSPORT_SEND_SUCCESS 
```

Does that mean there is an available event handler to retrieve the success/fail? I see in esp_insights.c line 247 an insights_event_handler() function so I have some code like this but it's failing to catch the event

```
static void esp_insights_event(void* arg, esp_event_base_t event_base, int32_t event_id, void* event_data) {
  esp_insights_transport_event_data_t *data = event_data;

  if (event_base != INSIGHTS_EVENT) {
    return;
  }

  switch(event_id) {
    case INSIGHTS_EVENT_TRANSPORT_SEND_SUCCESS:
        ESP_LOGI(TAG, "Insights transport success");
      break;
    case INSIGHTS_EVENT_TRANSPORT_SEND_FAILED:
        ESP_LOGI(TAG, "Insights transport failed");
      break;
    default:
        ESP_LOGI(TAG, "Insights defaulted");
      break;
  }
}

 esp_event_handler_register(INSIGHTS_EVENT, ESP_EVENT_ANY_ID, &esp_insights_event, NULL);
```

vikramdattu, 2023-08-04:
Hello @antoniuschan99 thanks for reaching out. 

> Does that mean there is an available event handler to retrieve the success/fail? I see in esp_insights.c line 247 an insights_event_handler() function so I have some code like this but it's failing to catch the event

Since the events go through common event handler, it does get delivered to all the subscribers in theory. Is the observation same for all the events?

5 seconds delay is good but it is not always useful as the messages are sent periodically. with minimum interval of 30 seconds.
Please remove that particular delay and go to deep_sleep after the transport send success event is received. You should also put a timeout for waiting on this event as well and go to sleep anyway. (This avoids deep_sleep logic not triggering in case of n/w issues etc). On the next boot cycle older messages will get sent on priority anyway.

antoniuschan99, 2023-08-08:
Thanks for the reply. I am having trouble catching the event. Both in my implementation of the event handler and also in insights_event_handler in esp_insights.c when I have INSIGHTS_DEBUG_ENABLED in menuconfig. I don't see             ESP_LOGI(TAG, "Data send success, msg_id:%d.", data ? data->msg_id : 0); get logged in the console.

The gist of the code is:
```
esp_insights_config_t config = {
    .log_type = ESP_DIAG_LOG_TYPE_ERROR | ESP_DIAG_LOG_TYPE_WARNING | ESP_DIAG_LOG_TYPE_EVENT,
    .node_id = sensorPin,
    .auth_key = insights_auth_key_start,
  };

  esp_err_t ret = esp_insights_init(&config);

  if (ret != ESP_OK) {
    ESP_LOGE(TAG, "Failed to initialize ESP Insights, err:0x%x", ret);

    return;
  }

  ret = esp_diag_heap_metrics_dump();

  if (ret != ESP_OK) {
    ESP_LOGE(TAG, "Failed to Dump Heap Metrics");
  }
}
```

And here are the logs.

```

I (2126) esp_netif_handlers: sta ip: 192.168.4.65, mask: 255.255.255.0, gw: 192.168.4.1
I (2126) WiFiConnect: Connected to IP Event Handler
I (2546) esp_rmaker_time: Initializing SNTP. Using the SNTP server: pool.ntp.org
I (2546) esp_rmaker_work_queue: Work Queue created.
I (2566) RTC_STORE: (write_at_offset): size 1025, available: 1025, filled 0, read_ptr 0, to_write 1
I (2566) RTC_STORE: (write_at_offset): size 1025, available: 1025, filled 0, read_ptr 0, to_write 8
I (2576) RTC_STORE: (write_at_offset): size 1025, available: 1025, filled 0, read_ptr 0, to_write 40
I (2586) RTC_STORE: before write_complete, filled 0, size 1025, read_offset 0, len 49
I (2596) RTC_STORE: after write_complete, filled 49, size 1025, read_offset 0, len 49
I (3216) esp_insights: =========================================
I (3226) esp_insights: Insights enabled for Node ID XXXXX
I (3236) esp_insights: =========================================
I (3236) RTC_STORE: (write_at_offset): size 2048, available: 2048, filled 0, read_ptr 0, to_write 1
I (3246) RTC_STORE: (write_at_offset): size 2048, available: 2048, filled 0, read_ptr 0, to_write 120
I (3256) RTC_STORE: before write_complete, filled 0, size 2048, read_offset 0, len 121
I (3266) RTC_STORE: after write_complete, filled 121, size 2048, read_offset 0, len 121
W (3236) insights_transport: connect callback not set
I (3286) esp_rmaker_work_queue: RainMaker Work Queue task started.
I (3296) esp_insights: Scheduling Insights timer for 60 seconds.
I (3306) esp_insights: Insights metrics metadata changed
I (3306) esp_insights: Insights meta data length 1099
```

antoniuschan99, 2023-08-09:
Upon looking further, it might be because I'm using HTTP instead of MQTT?

I see a RMAKER_MQTT_EVENT_PUBLISHED in rmaker_common_event_handler in esp_insights_mqtt.c. Which then does a             esp_event_post(INSIGHTS_EVENT, INSIGHTS_EVENT_TRANSPORT_SEND_SUCCESS, &data, sizeof(data), portMAX_DELAY);

vikramdattu, 2023-08-09:
@antoniuschan99 thanks for diving deeper. That's indeed the case.
Please apply attached change, and see if this satisfies your requirement.
[http_tport_status_events.patch](https://github.com/espressif/esp-insights/files/12298323/http_tport_status_events.patch)

antoniuschan99, 2023-08-09:
haha great that's where I'm at actually. Was just able to see the HTTP_EVENT_ since my log level was Info instead of Debug. Ok, will try tomorrow and report back, should be good since I'm seeing "HTTP_EVENT_ON_FINISH get logged.

closed 2023-08-09 by antoniuschan99