espressif/esp-insights #32: Upload firmware package .zip with API?

ftab, 2023-09-05:
Is it possible to automate the uploading of the firmware package generated at build time? It would be helpful if our CI/CD could do that during the deploy process so we don't have to manually do it.

shahpiyushv, 2023-09-06:
@ftab , since ESP Insights and ESP RainMaker have a common login, you can refer to the RainMaker [user management APIs](https://swaggerapis.rainmaker.espressif.com/#/User%20Management) to get the refresh and access tokens, which can then be used for Insights APIs. To upload an image you can check out the [OTA Image Upload Request](https://swaggerapis.rainmaker.espressif.com/#/OTA%20Service/OTAImageUploadRequestURL) and [OTA Image Upload Confirm](https://swaggerapis.rainmaker.espressif.com/#/OTA%20Service/OTAImageUploadConfirmURL) APIs.

You can also refer to the [RainMaker CLI](https://github.com/espressif/esp-rainmaker-cli) which has the user login and token management. You can extend it to upload the images to Insights.

benaveld, 2024-11-25:
@shahpiyushv, What url should we be using? 

I have tried using <https://api.insights.espressif.com> but it only responds with `{'status': 'failure', 'description': 'Method not allowed'}`.
Using <https://api.rainmaker.espressif.com> or <https://cn5cgbin6k.execute-api.us-east-1.amazonaws.com/prod> URLs allows me to upload the package named `<project_name>-<fw_version>.zip` using the [OTA Image Upload Request](https://swaggerapis.rainmaker.espressif.com/#/OTA%20Service/OTAImageUploadRequestURL) and [OTA Image Upload Confirm](https://swaggerapis.rainmaker.espressif.com/#/OTA%20Service/OTAImageUploadConfirmURL) APIs. But then the firmware only shows up in the RainMaker dashboard and not as a firmware package in the Insights dashboard. 
I also tried using [OTA package upload](https://swaggerapis.rainmaker.espressif.com/#/OTA%20Service/adminuploadotapackage) API but it also result in the firmware only showing up in the Rainmaker dashboard.

In conclusion, using the RainMaker API does not allow us to upload the firmware package .zip to the Insights dashboard. So the question remains of how do we automate the uploading of the firmware package?

The following is my python script that i use for testing.

```python
import requests
import hashlib
import base64
import pprint

# base_url = "https://cn5cgbin6k.execute-api.us-east-1.amazonaws.com/prod/v1"
# base_url = "https://api.insights.espressif.com/v1"
base_url = "https://api.rainmaker.espressif.com/v1"
ota_url = base_url + "/admin/otaimage"
upload_request = ota_url + "/upload_request"

def get_md5_checksum_of_file(file_name: str):
    with open(file_name, "rb") as file_to_check:
        data = file_to_check.read()
        return base64.b64encode(hashlib.md5(data).digest())

firmware_path = "build/qesp32-v1.1.1.zip"

# login ------------------------------------------------------------------------------
r = requests.post(
    url=base_url + "/login2",
    json={"user_name": "", "password": ""},
)

r_json = r.json()
if r_json["status"] != "success":
    print("login failed")
    pprint.pp(r_json)
    exit(-1)

id_token = r_json["idtoken"]
access_token = r_json["accesstoken"]
refresh_token = r_json["refreshtoken"]

# request ota imag upload ------------------------------------------------------------------------------
image_type = "package"
md5_checksum = get_md5_checksum_of_file(firmware_path)
response = requests.get(
    headers={"Authorization": access_token},
    url=upload_request,
    params={
        "image_name": "qesp32-v1.1.1",
        "image_type": image_type,
        "md5_checksum": md5_checksum,
    },
)

r_json = response.json()
if r_json["status"] != "success":
    print("Failed request ota image upload")
    pprint.pp(r_json)
    exit(-2)

upload_url = r_json["upload_url"]
ota_image_id = r_json["ota_image_id"]

# upload ota imag ------------------------------------------------------------------------------
with open(firmware_path, "rb") as file:
    response = requests.put(
        url=upload_url,
        data=file,
        headers={
            "Content-type": "application/octet-stream",
            "Content-MD5": md5_checksum,
        },
    )

    # Check if the upload was successful
    if response.status_code != 200:
        print(f"Failed to upload with status code: {response.status_code}")
        exit(-3)

# confirm ota imag upload ------------------------------------------------------------------------------

response = requests.post(
    headers={
        "Authorization": access_token,
    },
    url=ota_url + "/upload_confirm",
    json={
        "ota_image_id": ota_image_id,
        "type": "Test-Type",
        "image_type": image_type,
    },
)
r_json = response.json()
if r_json["status"] != "success":
    print("Failed confirm ota image upload")
    pprint.pp(r_json)
    exit(4)

pprint.pp(r_json)
```

isthaison, 2025-01-15:
[image]

silicakes, 2025-04-03:
Hey, what's the status of this?

isthaison, 2025-04-03:
It's OK