espressif/esp-insights #34: Getting an Invalid Chip in Esp Insights Dashboard

antoniuschan99, 2023-09-30:
Hello, I have been able to get ESP Insights using OTA update so started testing flashing for production and am getting an 'Invalid Chip' error in the ESP Insights Web Portal under Nodes. I have these in sdkconfig:

CONFIG_IDF_TARGET_ARCH_XTENSA=y
CONFIG_IDF_TARGET_ARCH="xtensa"
CONFIG_IDF_TARGET="esp32"
CONFIG_IDF_TARGET_ESP32=y

CONFIG_ESP_INSIGHTS_ENABLED=y
CONFIG_ESP_INSIGHTS_COREDUMP_ENABLE=y

antoniuschan99, 2023-10-02:
looks like I had to perform: idf.py -DIDF_TARGET=esp32 reconfigure. I think every time you delete the build folder (or fullclean) the env that sets idf_target gets erased. CONFIG_IDF_TARGET="esp32" doesn't seem to work even though during the build it will say IDF_TARGET is not set, guessed 'esp32' from sdkconfig

antoniuschan99, 2023-10-02:
Upon investigating looks like on initial call on this block of code it gives the 'invalid chip' error.

However, on the subsequent calls it gets fixed and the Chip shows as ESP32, Project and Chip version also gets populated.

 char* sensorPin = get_sensor_pin();

  esp_insights_config_t config = {
    .log_type = ESP_DIAG_LOG_TYPE_ERROR | ESP_DIAG_LOG_TYPE_WARNING | ESP_DIAG_LOG_TYPE_EVENT,
    .node_id = sensorPin,
    .auth_key = insights_auth_key_start,
  };

  esp_err_t ret = esp_insights_init(&config);

vikramdattu, 2023-10-20:
@antoniuschan99 thanks for reporting. We will try to reproduce the issue and fix it.

Do you always get this issue once the node is up? Or is it a random issue?

antoniuschan99, 2023-10-23:
always getting it. I haven't looked into it further as it isn't a showstopper. If you need more info let me know!

vikramdattu, 2023-10-26:
Hi @antoniuschan99 as one of the very message after reboot OR on the first boot, sent by ESP-Insights is the boot time data which contains the chip_id.

BTW, can you please let me know if `char* sensor_pin = get_sensor_pin();` yields an unique ID and is consistent across reboots?

vikramdattu, 2023-10-26:
Hi @antoniuschan99 I am able to re-produce the behaviour. 

Here is what I see:
When node starts appearing on the dashboard, it shows the chip as Invalid Chip. 
This is just for few seconds and if you refresh the dashboard after, say 5 seconds, it'll show a valid chip.

This behaviour is because, the information is received by Insights' cloud in different message and it might take few seconds till it populates this information. This doesn't need the device to reboot in between, and I think you are mis-interpreting it as getting fixed after reboot.

To make sure, that you are seeing the same as the issue, can you please wait on `Invalid chip` page for few seconds and if the information changes after the page refresh? (without rebooting the device in between).

antoniuschan99, 2023-11-11:
ok just saw this. Will test and update you in next few days. thanks!

antoniuschan99, 2023-11-29:
I will close this issue for now!

closed 2023-11-29 by antoniuschan99