espressif/esp-insights #35: projbuild_gitconfig.py fails due to the empty _CMAKE_TOOLCHAIN_PREFIX argument

ImerHamidovic, 2023-10-20:
I have noticed that the _building_(_flashing_, _monitoring_) of the system **sometimes** fails due to the **empty** TOOLCHAIN_PREFIX argument.

Logs: 
```
Traceback (most recent call last):
  File "<path_to_the_script>/get_projbuild_gitconfig.py", line 25, in <module>
    TOOLCHAIN_PREFIX=sys.argv[6]
IndexError: list index out of range
```
It is hard to tell when and how this happens, but sometimes it just fails with the same error. I have noticed that issue won't appear once  _fullclean_ is performed before the build.

Looking at the _build.ninja_ file I have seen that the argument TOOLCHAIN_PREFIX does not exist once the build fails. 

I have tried to build the system using the idf.py directly as well as using the VS Code extension shortcuts. Both of them are sometimes failing. 

ESP-IDF version: 4.4.3

Thank you in advance,
Imer Hamidovic

ImerHamidovic, 2023-10-20:
It is my first time reporting some issue so I am not aware of how to properly do it. Please let me know if you need some more information.

vikramdattu, 2023-10-20:
Hi @ImerHamidovic,

When exactly do you encounter this issue? If toolchains are set properly (with `. ./export.sh`) and you can build the app, you should not get such error.

Did you update the asp-insights recently? If yes, you will need a clean build for this.

ImerHamidovic, 2023-10-20:
Here are some more logs:

`[1157/1159] Generating binary image from built executable
esptool.py v3.3.2
Creating esp32 image...
Merged 25 ELF sections
Successfully created esp32 image.
Generated <path_to_project_folderbuild/conboard.bin
[1158/1159] cd /......../Desktop/WorkingFolder/Schulthess/Projects/Connectivi...ess/Projects/ConnectivityBoard/conboard-esp32/conboard_root/build/conboard.bin
conboard.bin binary size 0x16f9c0 bytes. Smallest app partition is 0x1d0000 bytes. 0x60640 bytes (21%) free.
[1159/1159] Running utility command for app
FAILED: CMakeFiles/app.util 
cd <path_to_project_folderbuild && /....../.espressif/python_env/idf4.4_py3.10_env/bin/python <path_to_project_foldermanaged_components/espressif__esp_insights/scripts/get_projbuild_gitconfig.py <path_to_project_folder conboard HW_TEST_BLE_WORKING-54-g461e39b <path_to_project_folderbuild/project_build_config.json /.........../esp/esp-idf && /usr/bin/cmake -D BUILD_DIR=<path_to_project_folderbuild -D PROJECT_DIR=<path_to_project_folder -D PROJECT_NAME=conboard -D PROJECT_VER=HW_TEST_BLE_WORKING-54-g461e39b -D ARCHIVE_DIR=conboard-vHW_TEST_BLE_WORKING-54-g461e39b -D PROJ_CONFIG_FILE=project_build_config.json -D PARTITION_CSV_FILE=partition_table_16MB.csv -P <path_to_project_foldermanaged_components/espressif__esp_insights/scripts/gen_tar_dir.cmake && /usr/bin/cmake -E echo "===================== Generating insights firmware package build/conboard-vHW_TEST_BLE_WORKING-54-g461e39b.zip ======================" && /usr/bin/cmake -E tar cfv conboard-vHW_TEST_BLE_WORKING-54-g461e39b.zip conboard-vHW_TEST_BLE_WORKING-54-g461e39b --format=zip && /usr/bin/cmake -E remove_directory conboard-vHW_TEST_BLE_WORKING-54-g461e39b && /usr/bin/cmake -E remove project_build_config.json
Traceback (most recent call last):
  File "<path_to_project_foldermanaged_components/espressif__esp_insights/scripts/get_projbuild_gitconfig.py", line 25, in <module>
      ##TOOLCHAIN_PREFIX=sys.argv[6]
IndexError: list index out of range
ninja: build stopped: subcommand failed.`