espressif/esp-insights #36: ESP Insights simultaneous HTTP request error

connormartin12, 2023-11-18:
I've recently come across an issue where using esp insights, while also polling a separate HTTPS GET request, can cause issues on occasion. 

For example, I run a GET request every 5 seconds to check the active score of a hockey game. On occasion, insights will try to open an HTTPS request to upload to send the data to the cloud at the same time the HTTPS request for retrieving the score is open (or vice versa). 

This isn't a huge deal as insights will just try again later and succeed. However, on a VERY rare occasion, this will somehow create a loop of back to back http failures that flood the console with logs and render the esp useless until the program is restarted.

Sadly I don't have a minimum reproducible example as this is so rare I've only ever caught it 4 times now. If there's a way to possibly let insights know when it should not attempt to open a connection to send the logs, that would be great.

vikramdattu, 2023-11-19:
Hi @connormartin12 you might very well be running out of sockets in your scenario.
The insights requests are executed from `esp_rmaker_work_queue`. To avoid the scenario happening, you may simply call `esp_rmaker_work_queue_add_task` with the task you want to execute. This will make sure both of these requests go in serial and not parallel at any instance. 
Do check `esp_rmaker_work_queue.h` for API details.

connormartin12, 2023-11-19:
Hi @vikramdattu, thanks for the advice. I tried this and it successfully solved the issue I was experiencing.

closed 2023-11-19 by connormartin12