espressif/esp-insights #37: spam "wifi:Haven't to connect to a suitable AP now!" when using 4G or ethernet

TiDiChi, 2024-01-12:
Hi, i am learning about esp-insights. I want to add this into my project using 4G/Ethernet/Wifi. 
When i config my project using Wifi, my device runs normally, but if 4G/Ethernet this spams "wifi:Haven't to connect to a suitable AP now!". 
What should i do to fix this !

sarveshb14, 2024-01-15:
Hi @TiDiChi ,

This warning print `Haven't to connect to a suitable AP now!` is displayed when [esp_wifi_sta_get_ap_info(wifi_ap_record_t *ap_info)](https://github.com/espressif/esp-idf/blob/b3f7e2c8a4d354df8ef8558ea7caddc07283a57b/components/esp_wifi/include/esp_wifi.h#L560) is called before [esp_wifi_connect(void)](https://github.com/espressif/esp-idf/blob/b3f7e2c8a4d354df8ef8558ea7caddc07283a57b/components/esp_wifi/include/esp_wifi.h#L404).

You can ignore this message as application (`esp-insights` in this case) uses this API just to check whether station is connected to Wi-Fi or not. 

cc: @vikramdattu

TiDiChi, 2024-01-16:
> Hi @TiDiChi ,
>
> This warning print `Haven't to connect to a suitable AP now!` is displayed when [esp_wifi_sta_get_ap_info(wifi_ap_record_t [...]

Thank you very much. I did it and this spam did not appear!

vikramdattu, 2024-01-18:
@TiDiChi if you're using interface other than Wi-Fi, it is better to use the `netif` APIs to query if that particular interface is connected or not. That would be much cleaner.

e.g., for ethernet

```
     esp_netif_t *esp_netif = esp_netif_get_handle_from_ifkey("ETH_DEF"); // ethernet handle
     if (esp_netif_is_netif_up(esp_netif) && esp_netif_get_ip_info(esp_netif, &ip_info) == ESP_OK) {
         // Do your stuff here
     }
```

Do check `esp_netif.h`.