espressif/esp-insights #38: eFuse error when running on qemu-system-riscv32

yaghmr, 2024-03-01:
I am trying to run the `esp-insights` app for ESP32-C3 in QEMU using Espressif's `qemu-system-riscv32`.
After compiling it and creating the flash with `esptool.py` as mentioned here: [https://github.com/espressif/esp-toolch ... to-emulate](https://github.com/espressif/esp-toolchain-docs/blob/main/qemu/esp32c3/README.md#compiling-the-esp-idf-program-to-emulate), I get the error "_W (114) eFuse: calibration efuse version does not match, set default version to 0_" at startup, and the application freezes.

What could be causing this?

Harshal5, 2024-03-04:
Hello @yaghmr,

It looks like ADC self-calibration is yet to be supported in ESP32-C3 QEMU.
A quick fix for this would be burning the efuse `BLK_VERSION_MAJOR` ([reference](https://github.com/espressif/esp-idf/blob/master/components/efuse/esp32c3/include/esp_efuse_rtc_calib.h#L15)) by which you could bypass the self-calibration step.

Before running the application using QEMU, you need to burn the efuse `BLK_VERSION_MAJOR` using the following steps:
1. Start QEMU in download mode:
```
build/qemu-system-riscv32 -nographic \
    -machine esp32c3 \
    -drive file=flash_image.bin,if=mtd,format=raw \
    -global driver=esp32c3.gpio,property=strap_mode,value=0x02 \
    -drive file=qemu_efuse.bin,if=none,format=raw,id=efuse \
    -global driver=nvram.esp32c3.efuse,property=drive,value=efuse \
    -serial tcp::5555,server,nowait
```
2. Burn the efuse:
```
espefuse.py --before no_reset -p socket://localhost:5555 burn_efuse BLK_VERSION_MAJOR 1
```

Finally, you should be able to run the application.

yaghmr, 2024-03-14:
I ran `qemu-system-riscv32` and it says:

```
QEMU 8.1.3 monitor - type 'help' for more information
Adding SPI flash device
(qemu)
```

Then I launched `espefuse.py` and it says:

```
Reading updated efuses...
Checking efuses...
Successful
```

Now, what command should I use to start the application?

mKlemmOrg, 2024-04-24:
Hallo @yaghmr,
(and also others coming here):

the `qemu-system-riscv32` as provided by @Harshal5 boots into UART0 download mode (see [bootmodes](https://github.com/espressif/esp-toolchain-docs/blob/main/qemu/esp32c3/README.md#specifying-bootstrapping-mode)).

So a easy an easy way would be following the commands you already did, close your qemu [q].
(If you want to, you can have a look into qemu_fuse.bin, there should 0x3C04 be "1" now).
Now you can just start qemu again in normal boot mode referencing the qemu_efuse.bin

(you can also save it to not have to redo this process when rebuilding the project)

The command to start your application would be:
```
qemu-system-riscv32 -nographic \
    -machine esp32c3 \
    -drive file=flash_image.bin,if=mtd,format=raw \
    -global driver=esp32c3.gpio,property=strap_mode,value=0x08 \
    -drive file=qemu_efuse.bin,if=none,format=raw,id=efuse \
    -global driver=nvram.esp32c3.efuse,property=drive,value=efuse
```

However, you might have to do more adaptions (depending on your specific code) to get it run (was the case for me), but maybe it is one step further.

Harshal5, 2024-04-25:
Very sorry for the late response.

> Now, what command should I use to start the application?

As correctly mentioned by @mKlemmOrg, you need to boot `qemu-system-riscv32` in the boot mode using te following command:

```
qemu-system-riscv32 -nographic \
    -machine esp32c3 \
    -drive file=flash_image.bin,if=mtd,format=raw \
    -global driver=esp32c3.gpio,property=strap_mode,value=0x08 \
    -drive file=qemu_efuse.bin,if=none,format=raw,id=efuse \
    -global driver=nvram.esp32c3.efuse,property=drive,value=efuse
```