espressif/esp-insights #39: Missing menu point ESP-INSIGHTS

planeckatamb, 2024-03-28:
Hey, 

I am running a ESP32H2 Dev Board just bought a couple of days ago. 
With idf.py menuconfig I can run the configuration menu. 
Within Component config, there is no option for ESP-Insights. 
All the other menu points are there. 
Any ideas? Couldn't find any other issue on the internet with the same problem. 

BR
Mario

vikramdattu, 2024-03-29:
Hi @planeckatamb 
Which example are you running `idf.py menuconfig` for?

If the example which is not the part of this repo, you can still add insights to your project via component manager: https://components.espressif.com/components/espressif/esp_insights
You should now see the option under `idf.py menuconfig > Component Config >`

Add these lines to initialise the insights in your project.
https://github.com/espressif/esp-insights/blob/main/examples/minimal_diagnostics/main/app_main.c#L63-#L73

shripad621git, 2024-04-02:
@planeckatamb , for trying out the lighting-app esp32 you can refer the steps [here](https://github.com/project-chip/connectedhomeip/blob/master/examples/lighting-app/esp32/README.md) in connectedhomeip.

planeckatamb, 2024-04-10:
However, Wi-Fi is not available on the ESP32H2 so maybe also no ESP Insights. Nevertheless, I tried it with the ESP32C6 and it worked flawlessly. So I think it is a matter of hardware.

closed 2024-04-10 by planeckatamb