espressif/esp-insights #40: We could not find the firmware image to cross-reference this log entry. Please upload a firmware image to get cross-referenced information for this log entry.

filzek, 2024-05-08:
Despite resetting everything, performing an idf.py fullclean, and rebuilding, when we upload the zip package to the ESP-Insight Firmware Packages tab, it shows the firmware there. However, the error on the dashboard still states:

`wifi_metrics
wifi_metrics: string at ROData: 0x3f41ae78 PC: 0x400da949
0x400da949
We could not find the firmware image to cross-reference this log entry. Please upload a firmware image to get cross-referenced information for this log entry.
`

The node does show the correct firmware version, so there appears to be a problem with the dashboard's ability to conclude the analysis.

vikramdattu, 2024-05-09:
Hi @filzek older logs cannot be referenced from newer firmware package. As the firmware sha would be different for the firmware which sent the logs and the one you uploaded later.

New logs coming from the device (if you have also flashed the corresponding firmware on device) however will get resolved with the new package.

filzek, 2024-05-09:
It was done in this order, build flash and wait until package upload to cloud, but understood, maybe some wrong with the package, the new package is working as intent

closed 2024-05-09 by filzek