espressif/esp-insights #44: ESP Insights Fails

michaelboeding, 2024-08-06:
I (2063) esp_rmaker_work_queue: Work Queue created.
E (2073) esp_insights: Failed to register event handler for INSIGHTS_EVENTS
E (2083) esp_insights: Failed to enable ESP Insights.
W (2083) insights_transport: disconnect callback not set
I (2093) esp_rmaker_work_queue: esp_rmaker_work_queue was successfully deinitialized

When I try to setup esp insights in my project I get the above error - any suggestions?

shahpiyushv, 2024-08-08:
Seems your code may be missing the esp_event_loop_create_default() API call which is required before any event handlers can be registered. Check this and any other such missing calls by referring our [examples](https://github.com/espressif/esp-insights/blob/main/examples/minimal_diagnostics/main/app_main.c)

michaelboeding, 2024-08-12:
I call the esp_event_loop_create_default() when I setup wifi. I'm not using rainmaker could that have something to do with it?

vikramdattu, 2024-08-12:
@michaelboeding  did you try example from esp-insights (or from examples present in esp-raimaker if you're using RainMaker)? 
Does the example work fine and you get issues with your project?

michaelboeding, 2024-08-12:
Haven't I can try that. I'm not using rainmaker at all in my current project - besides whats used for insights.

vikramdattu, 2024-08-12:
Hi @michaelboeding please check the [README](https://github.com/espressif/esp-insights/blob/main/README.md).
Should be good enough as the starting point.