espressif/esp-insights #48: Build error in V5.1.2 (IDFGH-13761)

eme-bennytang, 2024-09-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Build without error

### What is the actual behavior?

Keep getting build error. But I havent changed anything in this linux machine, while I can build perfectly in May.

### Steps to reproduce.

# Default to 921600 baud when flashing and monitoring device
CONFIG_ESPTOOLPY_BAUD_921600B=y
CONFIG_ESPTOOLPY_BAUD=921600
CONFIG_ESPTOOLPY_COMPRESSED=y
CONFIG_ESPTOOLPY_MONITOR_BAUD_OTHER_VAL=115200
CONFIG_ESPTOOLPY_MONITOR_BAUD=1152000
CONFIG_ESPTOOLPY_FLASHSIZE_16MB=y
CONFIG_IDF_TARGET="esp32s3"

# main task
CONFIG_ESP_MAIN_TASK_STACK_SIZE=6144

#enable BT
CONFIG_BT_ENABLED=y
CONFIG_BT_NIMBLE_ENABLED=y

#disable BT connection reattempt
CONFIG_BT_NIMBLE_ENABLE_CONN_REATTEMPT=n

#enable lwip ipv6 autoconfig
CONFIG_LWIP_IPV6_AUTOCONFIG=y

# Use a custom partition table
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_PARTITION_TABLE_FILENAME="partitions.csv"
CONFIG_PARTITION_TABLE_OFFSET=0xC000

#enable lwIP route hooks
CONFIG_LWIP_HOOK_IP6_ROUTE_DEFAULT=y
CONFIG_LWIP_HOOK_ND6_GET_GW_DEFAULT=y

# Button
CONFIG_BUTTON_PERIOD_TIME_MS=20
CONFIG_BUTTON_LONG_PRESS_TIME_MS=5000

# disable softap by default
CONFIG_ESP_WIFI_SOFTAP_SUPPORT=n

# Disable DS Peripheral
CONFIG_ESP_SECURE_CERT_DS_PERIPHERAL=n

# Default SN to 1234567890
CONFIG_USE_TEST_SERIAL_NUMBER="1234567890"

# Default to 1000Hz tick for FreeRTOS
CONFIG_FREERTOS_HZ=1000

# Use compact attribute storage mode
CONFIG_ESP_MATTER_NVS_USE_COMPACT_ATTR_STORAGE=y

# Build type
CONFIG_APP_REPRODUCIBLE_BUILD=y

# Default factory partition
CONFIG_CHIP_FACTORY_NAMESPACE_PARTITION_LABEL="nvs_factory"
CONFIG_CHIP_CONFIG_NAMESPACE_PARTITION_LABEL="nvs_factory"

# Default build matter OTA iamge
CONFIG_ENABLE_OTA_REQUESTOR=y
CONFIG_ENABLE_ENCRYPTED_OTA=y
CONFIG_OTA_AUTO_REBOOT_ON_APPLY=y
CONFIG_OTA_AUTO_REBOOT_DELAY_MS=5000

# Enable HKDF in mbedtls
CONFIG_MBEDTLS_HKDF_C=y

# Bootloader log level
CONFIG_BOOTLOADER_LOG_LEVEL_ERROR=y

# Log output
CONFIG_LOG_COLORS=y
CONFIG_LOG_TIMESTAMP_SOURCE_RTOS=y

# Increase LwIP IPv6 address number to 6 (MAX_FABRIC + 1)
# unique local addresses for fabrics(MAX_FABRIC), a link local address(1)
CONFIG_LWIP_IPV6_NUM_ADDRESSES=6

# CHIP Config
CONFIG_DISPATCH_EVENT_LONG_DISPATCH_TIME_WARNING_THRESHOLD_MS=2000
CONFIG_CHIP_LOG_FILTERING=y
CONFIG_MAX_EVENT_QUEUE_SIZE=50

# DNS-SD advertisement (multi-admin)
CONFIG_ENABLE_COMMISSIONABLE_DEVICE_TYPE=y

# Define device type
CONFIG_DEVICE_TYPE=769

# WiFi Station Options
CONFIG_ENABLE_WIFI_STATION=y
CONFIG_WIFI_STATION_RECONNECT_INTERVAL=10000
CONFIG_MAX_SCAN_NETWORKS_RESULTS=10
CONFIG_WIFI_SCAN_COMPLETION_TIMEOUT=10000
CONFIG_WIFI_CONNECTIVITY_TIMEOUT=30000
# end of WiFi Station Options

# Disable chip shell
CONFIG_ENABLE_CHIP_SHELL=n
# Disable chip tests
CONFIG_BUILD_CHIP_TESTS=n

# WatchDog config
CONFIG_ESP_INT_WDT=y
CONFIG_ESP_INT_WDT_TIMEOUT_MS=300
CONFIG_ESP_INT_WDT_CHECK_CPU1=y
CONFIG_ESP_TASK_WDT_EN=y
CONFIG_ESP_TASK_WDT_INIT=y
CONFIG_ESP_TASK_WDT_PANIC=y
CONFIG_ESP_TASK_WDT_TIMEOUT_S=5
CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU0=y
CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU1=y

######################### CERT #############################
# Default VID & PID
CONFIG_DEVICE_VENDOR_ID=0x105E
CONFIG_DEVICE_PRODUCT_ID=0x44A9
# Use DAC Provider implementation which reads attestation data from secure cert partition
CONFIG_SEC_CERT_DAC_PROVIDER=y
# Enable some options which reads CD and other basic info from the factory partition
CONFIG_ENABLE_ESP32_FACTORY_DATA_PROVIDER=y
CONFIG_ENABLE_ESP32_DEVICE_INSTANCE_INFO_PROVIDER=y
CONFIG_FACTORY_COMMISSIONABLE_DATA_PROVIDER=y
CONFIG_FACTORY_DEVICE_INSTANCE_INFO_PROVIDER=y

######################### SECURE #############################
# Security features (Enable hardware boot with test mode)
CONFIG_SECURE_SIGNED_ON_BOOT=y
CONFIG_SECURE_SIGNED_ON_UPDATE=y
CONFIG_SECURE_SIGNED_APPS=y
CONFIG_SECURE_BOOT_V2_RSA_SUPPORTED=y
CONFIG_SECURE_BOOT_V2_PREFERRED=y
CONFIG_SECURE_BOOT_V2_RSA_ENABLED=y
CONFIG_SECURE_SIGNED_APPS_RSA_SCHEME=y
CONFIG_SECURE_BOOT=y
CONFIG_SECURE_BOOT_V2_ENABLED=y
CONFIG_SECURE_BOOT_BUILD_SIGNED_BINARIES=y
CONFIG_SECURE_BOOT_SIGNING_KEY="MFMTG_Dev_SecureBoot_Key_Private.pem"
CONFIG_SECURE_BOOT_INSECURE=y
CONFIG_SECURE_FLASH_ENC_ENABLED=y
CONFIG_SECURE_FLASH_ENCRYPTION_AES256=y
CONFIG_SECURE_FLASH_ENCRYPTION_MODE_DEVELOPMENT=y
CONFIG_SECURE_FLASH_HAS_WRITE_PROTECTION_CACHE=y

# Potentially insecure options
CONFIG_SECURE_BOOT_ALLOW_JTAG=y
CONFIG_SECURE_BOOT_V2_ALLOW_EFUSE_RD_DIS=y
CONFIG_SECURE_FLASH_UART_BOOTLOADER_ALLOW_ENC=y
CONFIG_SECURE_FLASH_UART_BOOTLOADER_ALLOW_CACHE=y
CONFIG_SECURE_FLASH_SKIP_WRITE_PROTECTION_CACHE=y
CONFIG_SECURE_FLASH_CHECK_ENC_EN_IN_APP=y
CONFIG_SECURE_INSECURE_ALLOW_DL_MODE=y

######################### DEVELOPMENT #############################
# Default STA connected to router
#CONFIG_DEFAULT_WIFI_SSID="ASUS_N12HP3"
#CONFIG_DEFAULT_WIFI_PASSWORD="emehk2022"

# Log output
CONFIG_LOG_DEFAULT_LEVEL_INFO=y
CONFIG_LOG_DEFAULT_LEVEL=3
CONFIG_LOG_MAXIMUM_LEVEL_VERBOSE=y
CONFIG_LOG_MAXIMUM_LEVEL=5

# Use USB only for log
CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y
CONFIG_ESP_CONSOLE_SECONDARY_NONE=y
CONFIG_ESP_CONSOLE_MULTIPLE_UART=y
CONFIG_ESP_CONSOLE_UART_NUM=-1

# Use UART only for the log
#CONFIG_ESP_CONSOLE_UART_DEFAULT=y
#CONFIG_ESP_CONSOLE_SECONDARY_NONE=y
#CONFIG_ESP_CONSOLE_UART=y
#CONFIG_ESP_CONSOLE_MULTIPLE_UART=y
#CONFIG_ESP_CONSOLE_UART_NUM=0
#CONFIG_ESP_CONSOLE_UART_BAUDRATE=1152000

######################### PRODUCTION ###########################
# Log output
#CONFIG_LOG_DEFAULT_LEVEL_NONE=y
#CONFIG_LOG_DEFAULT_LEVEL=0
#CONFIG_LOG_MAXIMUM_EQUALS_DEFAULT=y
#CONFIG_LOG_MAXIMUM_LEVEL=0

# Console
#CONFIG_ESP_CONSOLE_NONE=y
#CONFIG_ESP_CONSOLE_SECONDARY_NONE=y
#CONFIG_ESP_CONSOLE_MULTIPLE_UART=y
#CONFIG_ESP_CONSOLE_UART_NUM=-1

### Build or installation Logs.

```plain
CMake Error at build/esp-idf/espressif__esp_insights/CMakeFiles/git-data/grabRef.cmake:48 (file):
  file failed to open for reading (No such file or directory):

    /home/benny/esp/workspace/MFMTG_SRC01_VXX/build/esp-idf/espressif__esp_insights/CMakeFiles/git-data/head-ref
Call Stack (most recent call first):
  /home/benny/esp/esp-idf/tools/cmake/third_party/GetGitRevisionDescription.cmake:83 (include)
  /home/benny/esp/esp-idf/tools/cmake/third_party/GetGitRevisionDescription.cmake:93 (get_git_head_revision)
  managed_components/espressif__esp_insights/CMakeLists.txt:45 (git_describe)
```

shahpiyushv, 2024-09-24:
How exactly have you included esp-insights? Is it via component manager or by cloning the repository? If it is by cloning the repository, have you cloned it with submodules?

```
git clone --recursive https://github.com/espressif/esp-insights.git
```

Note the --recursive option. This is required to pull in the various dependencies into esp-insights. In case you have already cloned the repository without this option, execute this to pull in the submodules: `git submodule update --init --recursive`

eme-bennytang, 2024-09-25:
Hi @shahpiyushv 
I have found the issue, somehow there is a .git file hidden inside my project folder, which introduced the error.
No problem building the project after removing the hidden file.

Thanks and I will close the issue.

closed 2024-09-25 by eme-bennytang

shahpiyushv, 2024-09-25:
The .git folder is auto created when you clone a repository using git commands. It is not meant to be deleted :)

eme-bennytang, 2024-09-25:
Hi @shahpiyushv 
I mean the .git file inside MY PROJECT, not inside the esp-idf/esp-matter
So I think its fine?
Thanks