espressif/esp-insights #49: Breaking backwards compatible removal of #define SHA_SIZE

SCCatADT, 2024-10-14:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I expected it to be able to compile esp-matter project without error.
Compiling using 
1. esp-idf v5.2.1 commit a322e6bdad4b6675d4597fb2722eea2851ba88cb
2. esp-matter commit c5d583525116bea8ce97e3567bf6f4e18985a960

### Actual behavior (suspected bug)

Instead it trigger 'SHA_SIZE' undeclared error

### Error logs or terminal output

```shell
managed_components/espressif__esp_insights/src/esp_insights_cbor_encoder.c:176:22: error: 'SHA_SIZE' undeclared here (not in a function)
  176 |     char sha_sum[2 * SHA_SIZE + 1]
```

### Steps to reproduce the behavior

1. remove managed_component folder
2. remove dependencies.lock
3. idf.py fullclean
4. idf.py set-target esp32s3
5. idf.py build

### Project release version

v1.2.1

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Ubuntu 22.04

### Shell

Bash

SCCatADT, 2024-10-14:
@vikramdattu , I've increase the version for esp_insights in `esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/idf_component.yml` to 1.2.1 and it solve the problem.

Will this be reflected on the esp-matter repo in the near future?

shripad621git, 2024-10-15:
@SCCatADT , the fix by pinning the version to 1.0.1 for espressif/esp_diag_data_store is available on latest esp-matter [here](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/idf_component.yml).

closed 2024-12-27 by shahpiyushv