espressif/esp-insights #52: 是否会开放esp-insights协议或部署服务端私有化部署？

CNiot, 2025-01-08:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

esp-insights 是非常好的理念和机制。但商业项目中需要严格控制数据范围，为此想咨询esp-insights 相关技术基础、协议是否会开放？最终能私有化部署 esp-insights 服务端，或基于开放协议的服务端。

shahpiyushv, 2025-01-20:
The specifications of esp-insights are not open, but that's just a procedural thing and we can open it up for better understanding and review. For private Insights deployment, you can get in touch with esp-rainmaker-admin@espressif.com or sales@espressif.com

CNiot, 2025-01-26:
理解了，非常感谢您的回应。

closed 2025-01-26 by CNiot